damiafuentes/DJITelloPy #105: DJITelloPy with ROS Noetic on Ubuntu 20.04

Helge543, 2021-08-11:
I want to setup my tello drones with DJITelloPy and ROS Noetic on my Ubuntu 20.04 machine. Is this possible?

M4GNV5, 2021-08-12:
There [already exist many such ROS drivers](https://github.com/search?l=Python&q=tello+ROS&type=Repositories) for the Tello. But at first glance most of them do not seem to use the DJITelloPy library and thus might not have support for all features such as state packets or video streaming.

You can still simply use this library in a ROS file, but a better (i.e. more ROS-like) idea is implementing a ROS driver which publishes a couple of topics with state information, a camera feed and controls.
If you do implement such a thing, feel free to publish it on github and i will link to it in the readme, as ROS has come up several times. We could even merge it into this repo itself if you like.

Helge543, 2021-08-12:
Hi Jakob, thanks a lot for the information and for the link. I'm just thinking about what's best. I also saw [this](http://wiki.ros.org/tello_driver) driver from the official ROS website which was built on the top of the [TelloPy](https://github.com/hanyazou/TelloPy) framework but I think it's just working on ROS kinetic which "ONLY supports Wily (Ubuntu 15.10), Xenial (Ubuntu 16.04) and Jessie (Debian 8) for debian packages"

closed 2021-11-28 by M4GNV5

hildebrandt-carl, 2022-12-01:
I am using this in ROS Noetic. I simply installed DJITelloPy as a python package and then I can use it in any python node.