damiafuentes/DJITelloPy #106: Any way to get the position of the drone

KadhirV, 2021-08-29:
I have a Tello Edu and I am trying to plot the real-time relative position of the drone on a computer. How would I approach this?

M4GNV5, 2021-08-30:
The EDU has mission pad support. When mission pad detection is enabled and a mission pad is detected the drones position relative to the mission pad is available via the `get_mission_pad_distance_x` functions.
Also see the [official mission pad user guide](https://dl-cdn.ryzerobotics.com/downloads/Tello/Tello%20Mission%20Pad%20User%20Guide.pdf)

KadhirV, 2021-08-30:
> The EDU has mission pad support. When mission pad detection is enabled and a mission pad is detected the drones position relative to the mission pad is available via the `get_mission_pad_distance_x` [...]

Awesome. I have not tried this yet, but I am wondering if the drone has the be in the detection range in order for this to work. If so, is there a reliable way to use the `get_acceleration_x, y, and z` functions to make this work?

M4GNV5, 2021-09-01:
Using the acceleration for calculation the position is probably a bad idea. A small error in the acceleration results in an incorrect speed which lets the calculated position drift away.

In general when trying to measure something you have to make sure measurement errors do not add up over time, but rather are reset for each new measurement.

cemoke, 2021-10-20:
What about using velX, velY, height, and yaw to compute position estimation? It should be better than using acceleration and integrating twice. Also, I saw MVO (I guess which provides internal position data of Tello flight controller) in other libraries which parses log messages of the Tello. Do you have any ideas on how to implement it in this library as well?

Of course, mission pads would be more reliable but the other solution would be much more generic.

closed 2022-07-11 by M4GNV5