damiafuentes/DJITelloPy #107: error unknown command

WillDeveloped, 2021-09-01:
I keep getting an error when I try to run:

drone = tello.Tello()
drone.connect()
drone.set_video_fps("Tello.FPS_30")

The error I get is an unknown command. 
I've updated the firmware on the drone to the most up to date.

WillDeveloped, 2021-09-01:
Reading though the documentation, it looks like those command were introduced in sdk3.0. Since my drone doesn't understand them, I'm guessing it's still on sdk2. I'll pull the sdk version when I get home and see what's going on. Maybe the Tello app didn't update the firmware as indicated?

closed 2021-09-01 by WillDeveloped

M4GNV5, 2021-09-02:
Your code is wrong. You have to call

```python
drone.set_fps(tello.Tello.FPS_30)
```

or better directly import the Tello class:

```python
from djitellopy import Tello

drone = Tello()
drone.connect()

drone.set_fps(Tello.FPS_30)
```