damiafuentes/DJITelloPy #110: how to use backgroundframeread? video latency

NasrulhaqAbdullah, 2021-09-12:
I have issue such that video streaming does have latency. I've read that there is Backgroundframeread().frame which can be used to read the current frame. However how can I access that? I'm looking forward to any solution because I have no idea what to do 😭 thanks

M4GNV5, 2021-09-14:
This will probably not decrease the latency (see #77 and #87), but you can use `BackgroundFrameRead` in a separate thread like shown in the [record video](https://github.com/damiafuentes/DJITelloPy/blob/master/examples/record-video.py) example:

```python
from threading import Thread
from djitellopy import Tello

tello = Tello()

tello.connect()

tello.streamon()
frame_read = tello.get_frame_read()

def frameWorkerFunc():
    height, width, _ = frame_read.frame.shape

    while True:
        img = frame_read.frame
        # TODO: do something with the image
        time.sleep(1 / 30)

    video.release()

# start our function in a separate thread
worker = Thread(target=frameWorkerFunc)
worker.start()

# TODO: do something with the tello
```

WillDeveloped, 2021-09-15:
> I have issue such that video streaming does have latency. I've read that there is Backgroundframeread().frame which can be used to read the current frame. However how can I access that? I'm looking forward to any solution because I have no idea what to do 😭 thanks

Did this solve your latency issues?

M4GNV5, 2021-11-13:
According to @ehong-tl using pyAV instead of opencv should reduce latency. In my tests i couldnt make out a difference in latency (both opencv and pyAV looked very low to me).
The current upstream version on git includes ehongs pull request. Feel free to try it yourself, maybe you can see an improvement.

closed 2021-11-13 by M4GNV5