damiafuentes/DJITelloPy #111: set_video_direction function doesnt work

maashon, 2021-10-04:
I want to access the downward camera but the function set_video_direction(Tello.CAMERA_DOWNWARD) does not work for my drone.
this is the response :
Traceback (most recent call last):
  File "D:\Users\Parsa\Desktop\Study materials\thesis\tello tutorils\first.py", line 10, in <module>
    print(tello.send_command_with_return(Tello.CAMERA_DOWNWARD))
  File "D:\Users\Parsa\Desktop\Study materials\thesis\tello tutorils\venv\lib\site-packages\djitellopy\enforce_types.py", line 53, in wrapper
    check_types(spec, *args, **kwargs)
  File "D:\Users\Parsa\Desktop\Study materials\thesis\tello tutorils\venv\lib\site-packages\djitellopy\enforce_types.py", line 45, in check_types
    raise TypeError("Unexpected type for '{}' (expected {} but found {})"
TypeError: Unexpected type for 'command' (expected <class 'str'> but found <class 'int'>)
[INFO] tello.py - 437 - Send command: 'streamoff'
[WARNING] tello.py - 447 - Aborting command 'streamoff'. Did not receive a response after 7 seconds
[INFO] tello.py - 437 - Send command: 'streamoff'
[WARNING] tello.py - 447 - Aborting command 'streamoff'. Did not receive a response after 7 seconds
[INFO] tello.py - 437 - Send command: 'streamoff'
[WARNING] tello.py - 447 - Aborting command 'streamoff'. Did not receive a response after 7 seconds
Exception ignored in: <function Tello.__del__ at 0x000001CBB39BDAF0>
Traceback (most recent call last):
  File "D:\Users\Parsa\Desktop\Study materials\thesis\tello tutorils\venv\lib\site-packages\djitellopy\enforce_types.py", line 54, in wrapper
  File "D:\Users\Parsa\Desktop\Study materials\thesis\tello tutorils\venv\lib\site-packages\djitellopy\tello.py", line 1019, in __del__
  File "D:\Users\Parsa\Desktop\Study materials\thesis\tello tutorils\venv\lib\site-packages\djitellopy\enforce_types.py", line 54, in wrapper
  File "D:\Users\Parsa\Desktop\Study materials\thesis\tello tutorils\venv\lib\site-packages\djitellopy\tello.py", line 1008, in end
  File "D:\Users\Parsa\Desktop\Study materials\thesis\tello tutorils\venv\lib\site-packages\djitellopy\enforce_types.py", line 54, in wrapper
  File "D:\Users\Parsa\Desktop\Study materials\thesis\tello tutorils\venv\lib\site-packages\djitellopy\tello.py", line 599, in streamoff
  File "D:\Users\Parsa\Desktop\Study materials\thesis\tello tutorils\venv\lib\site-packages\djitellopy\enforce_types.py", line 54, in wrapper
  File "D:\Users\Parsa\Desktop\Study materials\thesis\tello tutorils\venv\lib\site-packages\djitellopy\tello.py", line 486, in send_control_command
  File "D:\Users\Parsa\Desktop\Study materials\thesis\tello tutorils\venv\lib\site-packages\djitellopy\enforce_types.py", line 54, in wrapper
  File "D:\Users\Parsa\Desktop\Study materials\thesis\tello tutorils\venv\lib\site-packages\djitellopy\tello.py", line 528, in raise_result_error
Exception: Command 'streamoff' was unsuccessful for 4 tries. Latest response:	'Aborting command 'streamoff'. Did not receive a response after 7 seconds'

maashon, 2021-10-10:
I tried to update the firmware but there is an error in the app.my drone's firmware version is old but when I try to update the firmware, it downloads version v02.05.01.17 and when I press update, the app gives me the prompt : latest firmware! and it wont update.I tried flushing the apps cahce and data, also removing the app and reinstalling again but it just doesnt work.

M4GNV5, 2021-10-15:
Do you have an Tello EDU or regular Tello? Maybe the new SDK commands only work on the EDU versions? I have not yet tried to verify this though.
You porbably need to contact Ryze support when you have trouble with the official app or the drone. I cannot really help you with that.

closed 2021-11-13 by M4GNV5