damiafuentes/DJITelloPy #128: Not getting the video feed

Rishit0210, 2021-12-14:
Hi all,
I am having a problem getting the video feed from my Tello it is showing-

Traceback (most recent call last):
  File "C:/Users/rishi/Downloads/DJITelloPy-master/DJITelloPy-master/examples/Try test 1.py", line 11, in <module>
    cv2.imshow('Live Video', frame_read.frame)
cv2.error: OpenCV(4.5.1) C:\Users\appveyor\AppData\Local\Temp\1\pip-req-build-oduouqig\opencv\modules\highgui\src\window.cpp:376: error: (-215:Assertion failed) size.width>0 && size.height>0 in function 'cv::imshow' 

Can anyone help me fix it?

Regards,
Rishit0210

Rishit0210, 2021-12-14:
[image]
This is a picture of the error which I am getting.

closed 2021-12-16 by Rishit0210

M4GNV5, 2021-12-19:
I assume you found a solution as you closed this issue?

Rishit0210, 2022-02-13:
Yes, actually my computer was blocking my issue so the program was not working.