damiafuentes/DJITelloPy #132: socket.error: [Errno 98] Address already in use

Chentao2000, 2022-01-24:
if __name__=='__main__':
    drone1 = my_tello.Tello('',8888,'192.168.1.81',8889)
    drone2 = my_tello.Tello('',8889,'192.168.1.82',8889)

Traceback (most recent call last):
  File "/home/tao/Desktop/TelloGo/src/tello_pkg/scripts/tello_main.py", line 102, in <module>
    drone2 = my_tello.Tello('',8889,'192.168.1.82',8889)
  File "/home/tao/Desktop/TelloGo/src/tello_pkg/scripts/my_tello.py", line 81, in __init__
    self.socket_state.bind((local_ip,8890))
  File "/usr/lib/python2.7/socket.py", line 228, in meth
    return getattr(self._sock,name)(*args)
socket.error: [Errno 98] Address already in use

Chentao2000, 2022-01-24:
self.socket.bind((local_ip, local_port))

Chentao2000, 2022-01-24:
I want to call two drones, but the IP is different. How can I change it?

Chentao2000, 2022-01-24:
> if **name**=='**main**': drone1 = my_tello.Tello('',8888,'192.168.1.81',8889) drone2 = my_tello.Tello('',8889,'192.168.1.82',8889)
> 
> Traceback (most recent call last): File "/home/tao/Desktop/TelloGo/src/tello_pkg/scripts/tello_main.py", line 102, in drone2 = my_tello.Tello('',8889,'192.168.1.82',8889) File "/home/tao/Desktop/TelloGo/src/tello_pkg/scripts/my_tello.py", line 81, in **init** self.socket_state.bind((local_ip,8890)) File "/usr/lib/python2.7/socket.py", line 228, in meth return getattr(self._sock,name)(*args) socket.error: [Errno 98] Address already in use

M4GNV5, 2022-01-25:
https://github.com/damiafuentes/DJITelloPy/issues/91#issuecomment-855464080

closed 2022-01-25 by M4GNV5