damiafuentes/DJITelloPy #133: invalid imu after calibration and in well lit room

aizafaraz, 2022-01-27:
Hi, so I'm programming a drone and whenever I run the code it takes off but doesn't execute the rest of the code. I get an invalid imu error. I researched about it and tried it again in a well lit room and after calibration, but it still doesn't work. So, I updated the drone's firmware and ran the code again; still doesn't work. What can I do to solve this? I'm using a Tello Drone and Pycharm to code it (python).

M4GNV5, 2022-04-23:
Hey, sorry for the late response.

Did you manage to solve your issue? There is a calibration process in the app, maybe that helps?

Dawn-usk-M, 2022-05-06:
Same problem, calibration does't work
Has anyone solved this?

aizafaraz, 2022-05-06:
It has specific lighting needs. Even if it’s too much or too less. I used
some sunlight along with normal lighting I had inside. You will have to
experiment with different lighting.

Dawn-usk-M, 2022-05-06:
Thanks very much, I tried it outside, and now everything is OK.

closed 2022-05-06 by M4GNV5

aizafaraz, 2022-10-11:
Yes. I was able to solve it. Thank you though.

Aiza