damiafuentes/DJITelloPy #136: No frame problem, 'blue-frames', heating and issues with streaming video via OpenCV

anton-pintya, 2022-02-28:
Hi, I'm at the problem, connected with streaming video via djitellopy. I've seen same problems in closed and opened isses, but nothing changed after hours of editing, testing, copying and pasting.
First of all, every time I launch `streamon()` I get these warnings:
```
non-existing PPS 0 referenced
non-existing PPS 0 referenced
decode_slice_header error
no frame!
non-existing PPS 0 referenced
non-existing PPS 0 referenced
decode_slice_header error
no frame!
non-existing PPS 0 referenced
non-existing PPS 0 referenced
decode_slice_header error
no frame!
```
Then, when I opening the stream in the window via `cv2.imshow('Video', img)` I see frames with changed colors:

[image]

Rather strange, according to the fact, that I haven't seen such problem in other topics (seems it's inverting colors, or what, because everything in my grey and white room became blue on this frames)
After some time I recieve:
```left block unavailable for requested intra mode
error while decoding MB 0 25, bytestream 667
left block unavailable for requested intra4x4 mode -1
error while decoding MB 0 12, bytestream 5782
```

In addition, drone was heating much and blinking with a red color (battery is full) after running stream, and after processing the code - switches off . Tryied various suggestions, but the latest code, that gave at list this result is:

```
import cv2
import time
from djitellopy import Tello

tello = Tello()
tello.connect()

tello.send_command_with_return("port 8890 11111")

tello.streamon()
frame_read = tello.get_frame_read()

time.sleep(3)

while True:
    cv2.imshow("Streaming", frame_read.frame)

    if cv2.waitKey(1) == 27:
        frame_read.stop()
        tello.streamoff()
        break
```

Before trying this I received problems like `Aborting 'streamoff'` because of absence of heartbeat (adding last three rows fixed it I think).
Tried to specify port via `tello.send_command_with_return("port 8890 11111")` - nothing changed, tried to switch of/on Windows Defender - same. Please, if someone knows what to do, or at list can guess something, I'll be glad to read and try.

Versions:
Windows 10,
Python 3.8.6,
DJITelloPy 2.4.0
OpenCV 4.5.5.62

M4GNV5, 2022-02-28:
Hmm i assume pillow fetches the image as RGB, while opencv expects it to be BGR?
You probably have to convert the colors from RGB (as in red-green-blue) to BGR (as in blue-green-red):
```python
# initialize drone and start streaming..

img = cv2.cvtColor(frame_read.frame, cv2.COLOR_RGB2BGR)

cv2.imshow("Image", img)
cv2.waitKey(0)
```

Regarding the output: That is the normal output produced by the h264 decoder. It happens during start and from time to time when frames are lost. See #65 

Regaring overheating: Yes the drone tends to get hot pretty quick while idling. This is because the CPU is running and producing heat, which is usually cooled by the motor airflow.
Simply letting the drone take off should help. Or you can try the new [turn_motor_on](https://github.com/damiafuentes/DJITelloPy/blob/master/djitellopy/tello.py#L543-L546) functionality, which should cool the drone using the motors, but not have enough trottle to take off.

anton-pintya, 2022-02-28:
> Hmm i assume pillow fetches the image as RGB, while opencv expects it to be BGR? You probably have to convert the colors from RGB (as in red-green-blue) to BGR (as in blue-green-red):
>
> ```python
> # [...]

anton-pintya, 2022-02-28:
@M4GNV5 , thanks for your quick answer!
Strange... after adding one row (as you've said) I got this error (as far as I understood, this message occurs because of the absence of frame for the first iteration of the while loop):
```
line 20, in <module>
    img = cv2.cvtColor(frame_read.frame, cv2.COLOR_RGB2BGR)
cv2.error: OpenCV(4.5.5) D:\a\opencv-python\opencv-python\opencv\modules\imgproc\src\color.cpp:182: error: (-215:Assertion failed) !_src.empty() in function 'cv::cvtColor'

[h264 @ 000001c2208eb640] non-existing PPS 0 referenced
[h264 @ 000001c2208eb640] non-existing PPS 0 referenced
[h264 @ 000001c2208eb640] decode_slice_header error
[h264 @ 000001c2208eb640] no frame!
```
Then while loop breaks stopping the code

But another fact is strange: 20 minutes passed after my last running the code, and now I receive another messages from Tello initialization:
```
Send command: command
Response: b'ok'
Send command: port 8890 11111
Response: b'unknown command: port'
Send command: streamon
Response: b'ok'
```
Previously they were formed differently, like `tello.py - 426 - command 'streamon' ...`
Fine...

M4GNV5, 2022-02-28:
The frame read needs some time before the first frame is successfully received. Thats why there is a `time.sleep` in the example code you posted in the original issue.

When the `port` command is unavailable you are probably running an old firmware version. However you also do not need that command for regular usage. You can remove the `set_network_ports` call from your code.

anton-pintya, 2022-03-01:
@M4GNV5, added `time.sleep`, and decided to print `frame_read.frame` in the loop to see, whether it is None or not:
```python
# program above

time.sleep(5)

while True:
    print(frame_read.frame)
    if not frame_read.frame is None:
        img = cv2.cvtColor(frame_read.frame, cv2.COLOR_RGB2BGR)
        cv2.imshow("Streaming", img)

    cv2.waitKey(1)

# ...
``` 
Output:
```
[h264 @ 000001f3e03cb640] non-existing PPS 0 referenced
[h264 @ 000001f3e03cb640] non-existing PPS 0 referenced
[h264 @ 000001f3e03cb640] decode_slice_header error
[h264 @ 000001f3e03cb640] no frame!
[h264 @ 000001f3e03cb640] non-existing PPS 0 referenced
[h264 @ 000001f3e03cb640] non-existing PNone
None
None
None
None
None
None
None
None
None
None
None
None
None
None
None
...
```

So it didn't received any frames even after minute of working.

M4GNV5, 2022-03-01:
I assume you are missing something in your initialization code. Maybe the `streamon`? Or the `port` command did succeed in the past and now your tello is sending on different ports.

Please post your full program, otherwise its just guessing what might be wrong. Also please try resetting your Tello (pressing power button 15 seconds if i recall correctly)

anton-pintya, 2022-03-01:
@M4GNV5, here is the full code. 
```python
import keyboard
import cv2
import time
from djitellopy import Tello

tello = Tello()
tello.connect()
tello.send_command_with_return("port 8890 11111")

tello.streamon()
frame_read = tello.get_frame_read()

time.sleep(1)

while True:
    print(frame_read.frame)
    if not frame_read.frame is None:
        img = cv2.cvtColor(frame_read.frame, cv2.COLOR_RGB2BGR)
        cv2.imshow("Streaming", img)

    cv2.waitKey(1)

    if keyboard.is_pressed('esc'):
        frame_read.stop()
        tello.streamoff()
        break
```
Received output:
```
Send command: command
Response: b'ok'
Send command: port 8890 11111
Response: b'unknown command: port'
Send command: streamon
Response: b'ok'
None
None
None
None
None
```
Going to try resetting Tello as you've said, let's see what changes

M4GNV5, 2022-03-01:
> `tello.send_command_with_return("port 8890 11111")`

you should remove that line

> `if not frame_read.frame is None:`

you should use `if frame_read.frame is not None`.
Not that it makes a big difference though.

You included a camera screenshot in the original issue. Did it work back then? What did you change?

anton-pintya, 2022-03-01:
@M4GNV5, I copied code from the original issue and it didn't work: the same problem of "no frame", then I got:
```
File "C:/.../dji_drone.py", line 43, in <module>
    cv2.imshow("Streaming", frame_read.frame)
cv2.error: OpenCV(4.5.5) D:\a\opencv-python\opencv-python\opencv\modules\imgproc\src\color.cpp:182: error: (-215:Assertion failed) !_src.empty() in function 'cv::cvtColor'

eader error
```
So, even original code doesn't work now. Version of the module haven't been changed. As I understood, `cv2` didn't receive image to show it via `cv2.imshow()`, but what's difference: I don't know. I noticed in second comment that output strings of Tello changed after I posted this issue, maybe this changes are connected with difference in code behavior.

anton-pintya, 2022-03-01:
Seems that I got rid of this problem: reinstalled `djitellopy==2.4.0` package, installed `djitellopy2==2.3` (don't ask me why, I don't know what I downloaded it too for, in some issue discussions it worked), put `tello.streamoff()` before running `tello.streamon()`, added `time.sleep(1)` and in while loop put if-condition.
For me this code works good enough, no matter the fact that Tello sometimes turns off (I suppose because of overheating, cause I haven't arm motor, as @M4GNV5 advised above):
```python
import cv2
import time
from djitellopy import Tello

tello = Tello()
tello.connect()

tello.streamoff()
tello.streamon()
frame_read = tello.get_frame_read()

time.sleep(1)

while True:
    img = frame_read.frame

    if img is not None:
        cv2.imshow("Video Streaming", img)

    if cv2.waitKey(1) == 27:
        frame_read.stop()
        tello.streamoff()
        break

```
P.S. Many thanks to @M4GNV5 for quick answering, understandable answers and for helping in fact! It's nice to be able to ask a question and get a good answer not after a year :)

closed 2022-03-01 by anton-pintya