damiafuentes/DJITelloPy #138: turn_motor_on falling down with Exception

dvmixalkin, 2022-03-11:
I have some problems with Tello methos which is called "turn_motor_on"

tello = Tello()
tello.connect()
tello.turn_motor_on()

Exception: Command 'motoron' was unsuccessful for 4 tries. Latest response:	'unknown command: motoron'
Could you explain me? And is it possible to control each motor velocity?

M4GNV5, 2022-03-11:
The `turn_motor_on` function requires the latest tello firmware. Please update your tellos firmware using the official tello app.

I do not think it is possible to control each motors velocity. The motor speed is controlled based on the onboard acceleration / angular rotation sensor and is mostly used for flight stabilisation. Setting single motors to different speed will basically make your drone flip and/or crash.

dvmixalkin, 2022-03-12:
drone has latest tellos firmware version(01.04.92.01), but probles still occures

M4GNV5, 2022-07-11:
What drone are you using? Tellor or Tello EDU? What firmware version are you running? Maybe the new commands are only available on Tello EDU drones(?)

It is included in the [SDK 3.0 Docs](https://dl.djicdn.com/downloads/RoboMaster+TT/Tello_SDK_3.0_User_Guide_en.pdf)

mehdinafkha, 2022-07-11:
> What drone are you using? Tellor or Tello EDU? What firmware version are you running? Maybe the new commands are only available on Tello EDU drones(?)
> 
> It is included in the [SDK 3.0 Docs](https://dl.djicdn.com/downloads/RoboMaster+TT/Tello_SDK_3.0_User_Guide_en.pdf)

I am using the standrad version of Tello with a 01.04.92.01 WHICH IS THE LATEST FIRMARE VERSION, 

You are right, i think the Tello model is not compatible with the SDK 3.0
Thanks

closed 2022-09-25 by M4GNV5