damiafuentes/DJITelloPy #140: Downvision camera issue

wdliu356, 2022-03-15:
In Tello EDU, we can get a video stream from the downward camera. However, when I'm using the same get-frame function as the front camera, the frame sometimes works inappropriately as in the figure below. The top part is the correct figure, however, there are useless green parts at the bottom of the figure. Anyone shares the same issue as I have? Really need some advice here
[image]

kev278, 2022-09-27:
how were you able to access the bottom camera?

JaPeterson1, 2022-12-24:
> how were you able to access the bottom camera?

drone.set_video_direction(drone.CAMERA_DOWNWARD)
This only works on Tello Edu and Talent series drones.

Liang-mt, 2024-02-05:
Hello, may I ask which part of the code you use to call the camera? Currently, I encounter errors when I call the camera using the code.

amntheap, 2024-05-04:
You can simply just crop the additional parts with opencv. basically the downward camera has 640 by 480 image output.
img = cv2.resize(img, [640, 480])

also you can rotate it with this line of code:
img = cv2.rotate(img, cv2.ROTATE_90_CLOCKWISE)