damiafuentes/DJITelloPy #143: client socket bind to CONTROL_UDP_PORT?

jtamaki, 2022-04-08:
Why do you explicitly bind the client socket to CONTROL_UDP_PORT?https://github.com/damiafuentes/DJITelloPy/blob/master/djitellopy/tello.py#L113

I am trying to get a program using DJITelloPy to communicate with a simulator running on the same machine.
The simulator process also uses the same port, so I get an error.

I think the client socket should use the port assigned by os.
My program worked even after removing the relevant line.

M4GNV5, 2022-04-11:
I always assumed the Tello sends UDP packets back to this specific port, but to the IP address it received the `command` command from. I never tested wether it works with a different port though.
You should be able to try another port by setting `Tello.CONTROL_UDP_PORT = 8887` before initializing the first Tello instance.
Can you confirm that works with regular Tellos?

Also what simulator are you using? Because I am not aware of any Tello simulators in existence :smiley:

jtamaki, 2022-04-12:
> I always assumed the Tello sends UDP packets back to this specific port, but to the IP address it received the command command from.

Yes. I tried it with my "Tello(non edu)"

> You should be able to try another port by setting Tello.CONTROL_UDP_PORT = 8887 before initializing the first Tello instance.
> Can you confirm that works with regular Tellos?

It did not work because the program would try to send commands to port 8887 of "tello".
I think it is reasonable to remove the code that binds the client socket to the CONTROL_UDP_PORT port.

> Also what simulator are you using? Because I am not aware of any Tello simulators in existence 😃

I am trying to control [AirSim ](https://microsoft.github.io/AirSim/)multirotor by creating my own bridge program to communicate with Tello.

closed 2022-05-13 by M4GNV5