damiafuentes/DJITelloPy #145: Has anyone solved the background frame latency for video stream?

ZA0068, 2022-04-17:
I checked three threads that point out this issue, but the problem is that no one found a viable or even a good solution to this problem.

Has anyone tried to solve this issue?
I am looking for an example/implementation of this solution.

One example is using an AV stream instead of OpenCV to reduce latency, but I know nothing about the usage of AV.

Another Library is [hanyazou's Tellopy library](https://github.com/hanyazou/TelloPy), but it lacks many vital features that I need. Features like velocity measurements, altitude measurements,s, etc.

Please, I need some real solutions and not just suggestions.

NikEyX, 2022-04-21:
I haven't tried it yet (didn't have the time), but I saw that one of the PRs includes PyAV: https://github.com/damiafuentes/DJITelloPy/pull/120

This is actually reflected in the current github release, but doesn't seem to be the case for pip - that's still using OpenCV. Perhaps give it a try (clone this lib directly from github) and see if it improves?

NikEyX, 2022-04-23:
I can confirm that it works with PyAV as per this [PR](https://github.com/damiafuentes/DJITelloPy/pull/120).

In addition to that, there are several things that I changed in order to improve performance:
1. Separate control out to threads, that way grabbing images works while calling `takeoff()` or `land()`
2. Get rid of OpenCV entirely and display the entire feed via PyGame - this makes a huge difference, as OpenCV is really not suited for continuous streams
3. For me there was no need to call `to_image()` and casting every single frame to numpy arrays (in the tello.py function `update_frame`, so I do that operation in my main code only when necessary (i.e. based on the amount of FPS you want to display). 

Overall that got the stream delay from ~1.2 seconds to ~100ms, which is absolutely feasible at this point. Note I don't use a signal extender either, which was reported to improve performance further.

M4GNV5, 2022-04-23:
Your results sound amazing. Doesnt the BackgroundFrameRead already start a seprarate thread for receiving video? So I assume the additional thread is only required for PyGame?

Would you mind posting a small pygame example? We could include that as a "minimal delay" example in the repo.

I assume removing the `to_image` call has little to no effect on the stream delay, does it? It might consume some memory, but it probably makes it easier for users to work with data returned from the Tello library, so i guess we should keep it(?)

NikEyX, 2022-04-23:
I made a PR with the example that I used. It's possible that `to_image()` and casting has little effect, but for my purposes I only need like 6 frames per second (but as close to real time as possible), so I'd rather have it not spend any time on that operation.

The PR is quite similar to the example that's already in place, but improves on some issues that I faced when working with the examples (it really doesn't help that everyone suggests to use OpenCV's `imshow`).

ZA0068, 2022-05-31:
> 

OpenCV is mandetory and essensial for my project, I just can't ditch it.

M4GNV5, 2022-07-11:
You can still use OpenCV for image processing, but using pyAV for parsing the UDP stream to the tello reduces the delay. You get a regular numpy-array-image which you can then process using OpenCV