damiafuentes/DJITelloPy #149: Exception: Did not receive a state packet from the Tello

zelvet-velvet, 2022-05-10:
Hi,
I want to use djitellopy lib to controll my tello sdk 2.0 edu. I want to run some examples but I recieve the error below. 
↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓
[INFO] tello.py - 127 - Tello instance was initialized. Host: '192.168.10.1'. Port: '8889'.
[INFO] tello.py - 427 - Send command: 'command'
[INFO] tello.py - 451 - Response command: 'ok'
Traceback (most recent call last):
  File "test.py", line 5, in <module>
    tello.connect()
  File "/home/zel/DJITelloPy/djitellopy/enforce_types.py", line 54, in wrapper
    return func(*args, **kwargs)
  File "/home/zel/DJITelloPy/djitellopy/tello.py", line 536, in connect
    raise TelloException('Did not receive a state packet from the Tello')
djitellopy.tello.TelloException: Did not receive a state packet from the Tello
↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑
My system is windows 10 and I'm using WSL2 Ubuntu 18.04 with conda environment.
conda version:    4.8.3 
environment's python version:  3.8.3
I have tried to close the firewall but the result is still the same. 
Does anyone have any idea about how to solve the error? Thanks.

ch-MEIJIE, 2022-05-10:
You could try to turn off the fire-wall, it may block the link between drone and you pc.

zelvet-velvet, 2022-05-10:
I found out that I forgot to type False in tello.connect(False).

closed 2022-05-10 by zelvet-velvet