damiafuentes/DJITelloPy #152: Error running commands

jcerexhe, 2022-05-15:
Hey all. Having trouble with getting past Send command. So far I have: 
- installed dependencies
- connected to drone's wifi
- tried on mac and windows (output below)
- turned off firewalls

Py version: Python 3.9.6

Any thoughts? Thanks

```
python examples/simple.py
[ERROR] tello.py - 157 - [WinError 10022] An invalid argument was supplied
[INFO] tello.py - 126 - Tello instance was initialized. Host: '192.168.10.1'. Port: '8889'.
[DEBUG] tello.py - 423 - Waiting 0.0020008087158203125 seconds to execute command: command...
[INFO] tello.py - 426 - Send command: 'command'
Traceback (most recent call last):
  File "C:\Users\jcere\Documents\apps\DJITelloPy\DJITelloPy-master\examples\simple.py", line 8, in <module>
    tello.connect()
  File "c:\users\jcere\documents\apps\djitellopy\djitellopy-master\djitellopy\enforce_types.py", line 54, in wrapper
    return func(*args, **kwargs)
  File "c:\users\jcere\documents\apps\djitellopy\djitellopy-master\djitellopy\tello.py", line 523, in connect
    self.send_control_command("command")
  File "c:\users\jcere\documents\apps\djitellopy\djitellopy-master\djitellopy\enforce_types.py", line 54, in wrapper
    return func(*args, **kwargs)
  File "c:\users\jcere\documents\apps\djitellopy\djitellopy-master\djitellopy\tello.py", line 468, in send_control_command
    response = self.send_command_with_return(command, timeout=timeout)
  File "c:\users\jcere\documents\apps\djitellopy\djitellopy-master\djitellopy\enforce_types.py", line 54, in wrapper
    return func(*args, **kwargs)
  File "c:\users\jcere\documents\apps\djitellopy\djitellopy-master\djitellopy\tello.py", line 438, in send_command_with_return
    time.sleep(0.1)  # Sleep during send command
KeyboardInterrupt
```

jcerexhe, 2022-05-15:
Update:

Have experienced the connection working sometimes after first connecting to the drone's wifi. This is an example output:

```
python simple.py
[INFO] tello.py - 126 - Tello instance was initialized. Host: '192.168.10.1'. Port: '8889'.
[INFO] tello.py - 426 - Send command: 'command'
[INFO] tello.py - 450 - Response command: 'ok'
[INFO] tello.py - 426 - Send command: 'takeoff'
[INFO] tello.py - 450 - Response takeoff: 'ok'
[INFO] tello.py - 426 - Send command: 'left 100'
[INFO] tello.py - 450 - Response left 100: 'error No valid imu'
[INFO] tello.py - 426 - Send command: 'left 100'
[INFO] tello.py - 450 - Response left 100: 'error No valid imu'
[INFO] tello.py - 426 - Send command: 'left 100'
[INFO] tello.py - 450 - Response left 100: 'error No valid imu'
Traceback (most recent call last):
  File "/Users/jamie/apps/DJITelloPy/examples/simple.py", line 11, in <module>
    tello.move_left(100)
  File "/Users/jamie/apps/DJITelloPy/djitellopy/enforce_types.py", line 54, in wrapper
    return func(*args, **kwargs)
  File "/Users/jamie/apps/DJITelloPy/djitellopy/tello.py", line 626, in move_left
    self.move("left", x)
  File "/Users/jamie/apps/DJITelloPy/djitellopy/enforce_types.py", line 54, in wrapper
    return func(*args, **kwargs)
  File "/Users/jamie/apps/DJITelloPy/djitellopy/tello.py", line 605, in move
    self.send_control_command("{} {}".format(direction, x))
  File "/Users/jamie/apps/DJITelloPy/djitellopy/enforce_types.py", line 54, in wrapper
    return func(*args, **kwargs)
  File "/Users/jamie/apps/DJITelloPy/djitellopy/tello.py", line 475, in send_control_command
    self.raise_result_error(command, response)
  File "/Users/jamie/apps/DJITelloPy/djitellopy/enforce_types.py", line 54, in wrapper
    return func(*args, **kwargs)
  File "/Users/jamie/apps/DJITelloPy/djitellopy/tello.py", line 517, in raise_result_error
    raise TelloException("Command '{}' was unsuccessful for {} tries. Latest response:\t'{}'"
djitellopy.tello.TelloException: Command 'left 100' was unsuccessful for 4 tries. Latest response:	'error No valid imu'
[INFO] tello.py - 426 - Send command: 'land'
[WARNING] tello.py - 436 - Aborting command 'land'. Did not receive a response after 7 seconds
[INFO] tello.py - 426 - Send command: 'land'
[WARNING] tello.py - 436 - Aborting command 'land'. Did not receive a response after 7 seconds
[INFO] tello.py - 426 - Send command: 'land'
[WARNING] tello.py - 436 - Aborting command 'land'. Did not receive a response after 7 seconds
jamie@jc-laptop examples % python simple.py
[INFO] tello.py - 126 - Tello instance was initialized. Host: '192.168.10.1'. Port: '8889'.
[INFO] tello.py - 426 - Send command: 'command'
```

zenz, 2022-06-12:
My situation is that after takeoff, the tello drone lost controll, rest of the commands not executed.

M4GNV5, 2022-06-13:
@jcerex your second output looks like your drone needs calibration, please run the calibration program using the official app.
@zenz what exact error message are you getting? What does the drone replies to the commands? Please open an issue with the full output of your program.

closed 2022-06-13 by M4GNV5