damiafuentes/DJITelloPy #153: av.error.ExitError & error:"Failed to grab video frames from video stream"

zelvet-velvet, 2022-05-17:
My set up is wsl2 ubuntu conda environment with Python 3.6.13.

I am trying to test if my computer can recieve the video output from tello.
But it seems something go wrong.

code:
```

import cv2
import time
from djitellopy import Tello

tello = Tello()
tello.connect(False)

tello.streamon()
while True:
	img = tello.get_frame_read().frame
	cv2.imshow("Image",img)
	cv2.waitKey(1)

```

output :
```
[INFO] tello.py - 127 - Tello instance was initialized. Host: '192.168.10.1'. Port: '8889'.
[INFO] tello.py - 427 - Send command: 'command'
[INFO] tello.py - 451 - Response command: 'ok'
[INFO] tello.py - 427 - Send command: 'streamon'
[INFO] tello.py - 451 - Response streamon: 'ok'
Traceback (most recent call last):
  File "/home/zel/DJITelloPy/djitellopy/tello.py", line 1031, in __init__
    self.container = av.open(self.address, timeout=(Tello.FRAME_GRAB_TIMEOUT, None))
  File "av/container/core.pyx", line 355, in av.container.core.open
  File "av/container/core.pyx", line 226, in av.container.core.Container.__cinit__
  File "av/container/core.pyx", line 258, in av.container.core.Container.err_check
  File "av/error.pyx", line 336, in av.error.err_check
av.error.ExitError: [Errno 1414092869] Immediate exit requested: 'udp://0.0.0.0:11111'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "stream_test.py", line 13, in <module>
    img = tello.get_frame_read().frame
  File "/home/zel/DJITelloPy/djitellopy/enforce_types.py", line 54, in wrapper
    return func(*args, **kwargs)
  File "/home/zel/DJITelloPy/djitellopy/tello.py", line 410, in get_frame_read
    self.background_frame_read = BackgroundFrameRead(self, address)
  File "/home/zel/DJITelloPy/djitellopy/tello.py", line 1033, in __init__
    raise TelloException('Failed to grab video frames from video stream')
djitellopy.tello.TelloException: Failed to grab video frames from video stream
[INFO] tello.py - 427 - Send command: 'streamoff'
```

Does any one have any idea how to solve this error.
Thanks.

kogehipo, 2022-07-04:
I have the same issue. I looked #90 but the solutions there did not work, then reached to #153. I am using Windows11, WSL2, Ubuntu20.04, Python3.9.13 under pyenv and virtualenv. The same code works on M1 Mac, Raspberry Pi, Jetson Nano. So, I believe the problem is caused by environment. Below is my requirements.txt.

av==9.2.0
distlib==0.3.4
djitellopy==2.4.0
filelock==3.6.0
numpy==1.23.0
opencv-python==4.6.0.66
Pillow==9.1.1
platformdirs==2.5.1
protobuf==3.19.4
six==1.16.0

laineeded, 2022-08-08:
I also have the same error on my main PC which is a arch system, however there is no problem on neither of my laptops (one runs arch and the other one windows 11). I also did not have any problem in a windows 10 kvm vm (hosted on my main PC by redirecting my wifi dongle). All ran python 3.10 under virtualenv using opencv-python 4.6.0.66. It was identical code to Murtazas drone programming course (The image capture part) . I have tried various versions of both python and opencv.

A1exan10er, 2022-12-25:
I'm also facing this problem.
Using Windows 10 and Ubuntu 22.04, both didn't get video feedback from drone.

Tried to set environment same as Murtazza's settings:
Python Interpretor 3.7
djitellopy 1.5
numpy 1.19.0
opencv-python 4.3.0.36
pip 19.0.3
pygame 1.9.6
setuptools 40.8.0 
Didn't solve the problem.

Hoiidman, 2023-01-15:
anyone found a solution?

A1exan10er, 2023-01-15:
> anyone found a solution?

I checked the official SDK demo code example from DJI ( https://github.com/dji-sdk/Tello-Python )
And they are using Python 2.7, so in the end use their code directly.

The problem here is kind of skipped instead of solved.

M4GNV5, 2023-06-09:
It's hard to help with all the environment problems coming up. Since its working for most people its usually a combination problem of opencv + python.
I am about to release a new djitellopy version, which now uses pyAV instead of opencv for receiving frames. I hope this will help with many of these problems. Please everyone try updating your djitellopy versions

closed 2023-06-09 by M4GNV5

ccjimmy777, 2023-06-18:
I have the same problem. Is there any solution now?

bzbsng, 2023-12-05:
I have the same problem. Is there any solution now?

M4GNV5, 2023-12-06:
Which `djitellopy` version are you using? Can you post your output (maybe in a new issue)?
Using the newest version 2.5 from pypi should solve all opencv problems.

str1ngth3ory, 2024-01-09:
I have been stuck at the exact same issue. I have tried [#90](https://github.com/damiafuentes/DJITelloPy/issues/90) but it's not working. So it seems to be an environment issue, do we have a recommended list of dependencies, and in this case, for Ubuntu 22.04? Much appreciated.