damiafuentes/DJITelloPy #156: How works send_rc_control() ?

spider-x, 2022-07-09:
Hello, 
    how works the send_rc_control() method? When i run the method, how long is it working? Or how many times it send the command? I just know it sets velocity in one direction. When the command is sent, how long is the velocity increased or decreased?

M4GNV5, 2022-07-11:
The `send_rc_control` function **sets** the speed of the Tello. It is usually called repeatedly with updated values fed from either an algorithm or user input. I.e. call the `send_rc_control` function every 1 second.

The [pygame example](https://github.com/damiafuentes/DJITelloPy/blob/master/examples/manual-control-pygame.py#L174) does that based on which keyboard keys are pressed.

spider-x, 2022-07-11:
Thank you for explanation.