damiafuentes/DJITelloPy #157: non-existing PPS 0 referenced

rafiqrana, 2022-07-21:
`DJITelloPy/examples/record-video.py` does not work on Ubuntu 20.04 focal (x86-64) and Windows 11.

I got the following console message:

```
[INFO] tello.py - 126 - Tello instance was initialized. Host: '192.168.10.1'. Port: '8889'.
[INFO] tello.py - 426 - Send command: 'command'
[INFO] tello.py - 450 - Response command: 'ok'
[INFO] tello.py - 426 - Send command: 'streamon'
[INFO] tello.py - 450 - Response streamon: 'ok'
non-existing PPS 0 referenced
non-existing PPS 0 referenced
decode_slice_header error
no frame!
non-existing PPS 0 referenced
non-existing PPS 0 referenced
decode_slice_header error
no frame!
non-existing PPS 0 referenced
non-existing PPS 0 referenced
decode_slice_header error
no frame!
non-existing PPS 0 referenced
non-existing PPS 0 referenced
decode_slice_header error
no frame!
non-existing PPS 0 referenced
non-existing PPS 0 referenced
decode_slice_header error
no frame!
non-existing PPS 0 referenced
non-existing PPS 0 referenced
decode_slice_header error
no frame!
non-existing PPS 0 referenced
non-existing PPS 0 referenced
decode_slice_header error
no frame!
non-existing PPS 0 referenced
non-existing PPS 0 referenced
decode_slice_header error
no frame!
non-existing PPS 0 referenced
non-existing PPS 0 referenced
decode_slice_header error
no frame!
non-existing PPS 0 referenced
non-existing PPS 0 referenced
decode_slice_header error
no frame!
non-existing PPS 0 referenced
non-existing PPS 0 referenced
decode_slice_header error
no frame!
non-existing PPS 0 referenced
non-existing PPS 0 referenced
decode_slice_header error
no frame!
non-existing PPS 0 referenced
non-existing PPS 0 referenced
decode_slice_header error
no frame!
non-existing PPS 0 referenced
non-existing PPS 0 referenced
decode_slice_header error
no frame!
non-existing PPS 0 referenced
non-existing PPS 0 referenced
decode_slice_header error
no frame!
non-existing PPS 0 referenced
non-existing PPS 0 referenced
decode_slice_header error
no frame!
non-existing PPS 0 referenced
non-existing PPS 0 referenced
decode_slice_header error
no frame!
non-existing PPS 0 referenced
non-existing PPS 0 referenced
decode_slice_header error
no frame!
non-existing PPS 0 referenced
non-existing PPS 0 referenced
decode_slice_header error
no frame!
non-existing PPS 0 referenced
non-existing PPS 0 referenced
decode_slice_header error
no frame!
non-existing PPS 0 referenced
non-existing PPS 0 referenced
decode_slice_header error
no frame!
non-existing PPS 0 referenced
non-existing PPS 0 referenced
decode_slice_header error
no frame!
non-existing PPS 0 referenced
non-existing PPS 0 referenced
decode_slice_header error
no frame!
[INFO] tello.py - 426 - Send command: 'takeoff'
[INFO] tello.py - 450 - Response takeoff: 'ok'
[INFO] tello.py - 426 - Send command: 'up 50'
[INFO] tello.py - 450 - Response up 50: 'ok'
[INFO] tello.py - 426 - Send command: 'ccw 360'
left block unavailable for requested intra mode
error while decoding MB 0 27, bytestream 6065
[INFO] tello.py - 450 - Response ccw 360: 'ok'
[INFO] tello.py - 426 - Send command: 'land'
[INFO] tello.py - 450 - Response land: 'ok'
[INFO] tello.py - 426 - Send command: 'streamoff'
[WARNING] tello.py - 436 - Aborting command 'streamoff'. Did not receive a response after 7 seconds
[INFO] tello.py - 426 - Send command: 'streamoff'
[WARNING] tello.py - 436 - Aborting command 'streamoff'. Did not receive a response after 7 seconds
[INFO] tello.py - 426 - Send command: 'streamoff'
[WARNING] tello.py - 436 - Aborting command 'streamoff'. Did not receive a response after 7 seconds```

M4GNV5, 2022-09-25:
It looks like there is some data receibed, but the decoder is not able to read it. Are you running the latest version of djitellopy which uses pyav? How did you install it?

kev278, 2022-09-26:
i tried pip install av but that does not work

JaPeterson1, 2022-12-24:
I get the video decode error for the first 5-10s after starting the video stream but afterwards it goes away and the video is received normally. 
It looks like the cause for the file crashing is a communications issue with the drone after the landing as the opencv warning doesn't crash the program. If you are doing any image processing you will want to have some check that the frame was properly grabbed. 

Also, make sure you install the package from the downloaded repository instead of from pip to make sure you have the latest pyav version.

M4GNV5, 2022-12-26:
I assume we can close this then.
Now that a lot of people have successfully tested the pyAV version we can also make a release pushing it to pypi.

closed 2022-12-26 by M4GNV5