damiafuentes/DJITelloPy #159: Issue with Installaton

Manoharmamidipaka, 2022-07-30:
Hello,
I'm trying to install using developer's method it shows error
root@kali:~/DJITelloPy# pip install -e .
Obtaining file:///root/DJITelloPy
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/root/DJITelloPy/setup.py", line 3, in <module>
        with open("README.md", "r", encoding="utf-8") as fd:
    TypeError: 'encoding' is an invalid keyword argument for this function
    
please look

M4GNV5, 2022-07-30:
What python version are you using?

Manoharmamidipaka, 2022-07-30:
Python 3.6.5rc1

M4GNV5, 2022-09-25:
I assume the `encoding` parameter to `open` was introduced in a later version. Why are you using an `rc` version anyways?
Updating your python should solve the issue.

closed 2022-09-25 by M4GNV5