damiafuentes/DJITelloPy #162: Fails to fetch Frame and State data inside a Docker Container

achillesrasquinha, 2022-09-07:
Hi,

I'm trying to run the DJITelloPy inside a docker container. My host network is connected with the TELLO network however on running even the basic examples, DJITelloPy fails to connect to the Video or State Socket Streams. It successfully connects with the client streams, however.

Any clues?

achillesrasquinha, 2022-09-07:
So it seems like when binding the server sockets within Docker, Docker's own mapped IPs arrive as responses. This doesn't necessarily exist within the `drones` object within `tello.py`. Since UDP is connectionless, there seems to be no way to validate that the data actually comes from the requested client.

M4GNV5, 2022-09-25:
Can you provide your Dockerfile as well as your docker command / docker-compose file?

What address do the packets appear to be coming from? Maybe you can adjust the `host` parameter to the correct address? But then probably the packets would be sent to the wrong address. You could try adding a second parameter to allow separate outgoing and incoming adresses.

amneshia, 2022-11-27:
This is my version of the docker-compose.yml. I have also tried commenting out some combination of options at times but it does not work:

```
version: "3"
services:      
  mosquito:
    build:
      context: ..
      dockerfile: Dockerfile
    working_dir: /workspace
    deploy:
      resources:
        reservations:
          devices:
            - capabilities: [gpu]
    volumes:
      - ../:/workspace
    network_mode: host
    ipc: host
    privileged: true
    security_opt:
      - seccomp:unconfined
    cap_add:
      - SYS_PTRACE
    stdin_open: true
    tty: true
    ports:
      - "8889:8889"
      - "8890:8890" 
      - "11111:11111"
```
I should aim to run the python script inside the docker's bash. The error I receive is this:

```
[INFO] tello.py - 122 - Tello instance was initialized. Host: '192.168.10.1'. Port: '8889'.
[INFO] tello.py - 437 - Send command: 'command'
[INFO] tello.py - 461 - Response command: 'ok'
Traceback (most recent call last):
  File "mosquito.py", line 5, in <module>
    mosquito.connect()
  File "/opt/conda/lib/python3.7/site-packages/djitellopy/enforce_types.py", line 54, in wrapper
    return func(*args, **kwargs)
  File "/opt/conda/lib/python3.7/site-packages/djitellopy/tello.py", line 546, in connect
    raise Exception('Did not receive a state packet from the Tello')
Exception: Did not receive a state packet from the Tello
```