damiafuentes/DJITelloPy #167: Not getting response packets for commands

JaPeterson1, 2022-11-25:
Whenever I call a command with a response expected, the command is properly sent to the drone but the "ok" response from the drone is not detected in the library. The response packet is being sent by the drone (I can see it with wireshark), and I turned off my firewall so it shouldn't be getting blocked. 

I am using a Tello EDU and using the library as downloaded from github and installed with developer mode. 

Code: 
`from djitellopy import Tello
import time
drone=Tello()
drone.connect()
drone.takeoff()
time.sleep(5)
drone.land()`

Expected output: drone will takeoff, hover for 5 seconds, and land. 

Actual output: djitellopy.tello.TelloException: Command 'command' was unsuccessful for 4 tries. Latest response:       'Aborting command 'command'. Did not receive a response after 7 seconds'

As shown below, the drone is sending back the "ok" response. 
[image]

Thank you in advance for your help!

JaPeterson1, 2022-11-25:
I tested this with the client socket bind code that was removed in issue #143 and is mentioned for being added back in #163. The code from #163 solves the issue.

IEEERoboticsWSU, 2022-11-26:
https://github.com/damiafuentes/DJITelloPy/pull/168 should fix this

closed 2023-01-02 by M4GNV5