damiafuentes/DJITelloPy #172: Setup.py giving syntax error.

hildebrandt-carl, 2023-01-02:
Hi I am busy working on another issue and I noticed that there is an issue with the `setup.py` file.

The install requires currently is:
```python
    install_requires=[
        'numpy',
        'opencv-python',
        'av'.
        'pillow'
    ],
```

However it should be
```python
    install_requires=[
        'numpy',
        'opencv-python',
        'av',
        'pillow'
    ],
```

I will submit a pull request that fixes this shortly.

IEEERoboticsWSU, 2023-01-02:
My bad

hildebrandt-carl, 2023-01-02:
No worries. Was a very easy fix. I am busy working on Issue #166, and found this bug.

hildebrandt-carl, 2023-01-03:
Whoops I submitted the pull request from the wrong account. Sorry about that. I have now re-submitted the pull request.

IEEERoboticsWSU, 2023-01-03:
may want to close this

hildebrandt-carl, 2023-01-03:
Just checked and it seems like the issue is still in the `setup.py`

M4GNV5, 2023-01-04:
should be fixed now, since #174 is merged

closed 2023-01-04 by M4GNV5