damiafuentes/DJITelloPy #177: Video stream can`t load while using tkinter

FenixCrafter, 2023-02-15:
I tried making a Drone ui with tkinter and cv2 but it doesnt work,

the error im getting is :
[h264 @ 000001d7c595c880] non-existing PPS 0 referenced
[h264 @ 000001d7c595c880] decode_slice_header error
[h264 @ 000001d7c595c880] no frame!

and

TypeError: Can't convert object to 'str' for 'filename'
[INFO] tello.py - 437 - Send command: 'land'

The code:

 from djitellopy import Tello
import cv2, math, time
from tkinter import* 
from tkinter import Button
from tkinter import *
from tkinter import ttk
from PIL import ImageTk, Image
import cv2

tello = Tello()
tello.connect()

tello.streamon()

root = Tk()
root.title("DroneUI")
frm = ttk.Frame(root, width=500, height=500)
frm.grid(row=0, column=0)

tello.takeoff()
while True:
    frame = tello.get_frame_read()
    img = frame.frame
    tello.BITRATE_AUTO

    print(img)
    cv2.imshow("drone",img)
    # blue,green,red = cv2.split(img)
    # imgMerged = cv2.merge((red,green,blue))
    image = cv2.imread(img)

    im = Image.fromarray(image)
    imgtk = ImageTk.PhotoImage(image=im)

    Label(frm, image=imgtk).pack()

root.mainloop()

jazztyan, 2023-02-28:
I just made some code to put the img from tello / cv2 to tkinter. Just try this code, please...

`from djitellopy import tello
import cv2
from tkinter import *
from tkinter import ttk
from PIL import ImageTk, Image

root=Tk()
drone=True #True if img source frome Tello or False if img source from default Webcam

if drone:
    me = tello.Tello()
    me.connect()
    me.streamon()
    transferimg = me.get_frame_read().frame
    transferimg = cv2.resize(transferimg, (360, 240))
else:
    cap = cv2.VideoCapture(0)
    success, transferimg = cap.read()
    transferimg = cv2.resize(transferimg, (360,240))

w, h = 360, 240

def myloop():
    if drone:
        img = me.get_frame_read().frame
    else:
        success, img = cap.read()

    img = cv2.resize(img, (w, h))

    imageFrame=ttk.Frame(root)
    imageFrame.grid(column=0, row=1)

    camIMG=ttk.Label(imageFrame)
    camIMG.grid(column=0, row=0)

    imgRGBA = cv2.cvtColor(img, cv2.COLOR_BGR2RGBA)
    img = Image.fromarray(imgRGBA)
    imgtk = ImageTk.PhotoImage(image=img)
    camIMG.imgtk = imgtk
    camIMG.configure(image=imgtk)

    root.after(10, myloop)

myloop()

root.mainloop()`

jazztyan, 2023-02-28:
After I tried that way, my laptop became slow. So I let cv2 and tkinter separate using threading method. I also using matplotlib for my project, You can see the result [here](https://youtu.be/gCIC_auXbbA)

closed 2025-05-28 by FenixCrafter