damiafuentes/DJITelloPy #181: Response error on Windows 11

GreatenAnoymous, 2023-04-09:
I tried running the "simple.py" in the example folder on Windows 11. The drone was connected and took off successfully. However, it had no response after sending the "move_left"  command. Here is the error message:

PS C:\work\tello> python .\simple.py
[INFO] tello.py - 122 - Tello instance was initialized. Host: '192.168.10.1'. Port: '8889'.
[INFO] tello.py - 437 - Send command: 'command'
[ERROR] tello.py - 457 - 'utf-8' codec can't decode byte 0xcc in position 0: invalid continuation byte
[INFO] tello.py - 437 - Send command: 'command'
[INFO] tello.py - 461 - Response command: 'ok'
[INFO] tello.py - 437 - Send command: 'takeoff'
[INFO] tello.py - 461 - Response takeoff: 'ok'
[INFO] tello.py - 437 - Send command: 'left 100'
[INFO] tello.py - 461 - Response left 100: 'error enter stop'
[INFO] tello.py - 437 - Send command: 'left 100'
[INFO] tello.py - 461 - Response left 100: 'error enter stop'
[INFO] tello.py - 437 - Send command: 'left 100'
[INFO] tello.py - 461 - Response left 100: 'error enter stop'
Traceback (most recent call last):
  File "C:\work\tello\test_tello.py", line 8, in <module>
    tello.move_left(100)
  File "C:\Users\gt\AppData\Local\Programs\Python\Python310\lib\site-packages\djitellopy\enforce_types.py", line 54, in wrapper
    return func(*args, **kwargs)
  File "C:\Users\gt\AppData\Local\Programs\Python\Python310\lib\site-packages\djitellopy\tello.py", line 635, in move_left
    self.move("left", x)
  File "C:\Users\gt\AppData\Local\Programs\Python\Python310\lib\site-packages\djitellopy\enforce_types.py", line 54, in wrapper
    return func(*args, **kwargs)
  File "C:\Users\gt\AppData\Local\Programs\Python\Python310\lib\site-packages\djitellopy\tello.py", line 614, in move
    self.send_control_command("{} {}".format(direction, x))
  File "C:\Users\gt\AppData\Local\Programs\Python\Python310\lib\site-packages\djitellopy\enforce_types.py", line 54, in wrapper
    return func(*args, **kwargs)
  File "C:\Users\gt\AppData\Local\Programs\Python\Python310\lib\site-packages\djitellopy\tello.py", line 486, in send_control_command
    self.raise_result_error(command, response)
  File "C:\Users\gt\AppData\Local\Programs\Python\Python310\lib\site-packages\djitellopy\enforce_types.py", line 54, in wrapper
    return func(*args, **kwargs)
  File "C:\Users\gt\AppData\Local\Programs\Python\Python310\lib\site-packages\djitellopy\tello.py", line 528, in raise_result_error
    raise Exception("Command '{}' was unsuccessful for {} tries. Latest response:\t'{}'"
Exception: Command 'left 100' was unsuccessful for 4 tries. Latest response:    'error enter stop'
[INFO] tello.py - 437 - Send command: 'land'
[WARNING] tello.py - 447 - Aborting command 'land'. Did not receive a response after 7 seconds
[INFO] tello.py - 437 - Send command: 'land'
[WARNING] tello.py - 447 - Aborting command 'land'. Did not receive a response after 7 seconds
[INFO] tello.py - 437 - Send command: 'land'
[WARNING] tello.py - 447 - Aborting command 'land'. Did not receive a response after 7 seconds
Exception ignored in: <function Tello.__del__ at 0x00000279FEFAB130>
Traceback (most recent call last):
  File "C:\Users\gt\AppData\Local\Programs\Python\Python310\lib\site-packages\djitellopy\enforce_types.py", line 54, in wrapper
  File "C:\Users\gt\AppData\Local\Programs\Python\Python310\lib\site-packages\djitellopy\tello.py", line 1019, in __del__
  File "C:\Users\gt\AppData\Local\Programs\Python\Python310\lib\site-packages\djitellopy\enforce_types.py", line 54, in wrapper
  File "C:\Users\gt\AppData\Local\Programs\Python\Python310\lib\site-packages\djitellopy\tello.py", line 1006, in end
  File "C:\Users\gt\AppData\Local\Programs\Python\Python310\lib\site-packages\djitellopy\enforce_types.py", line 54, in wrapper
  File "C:\Users\gt\AppData\Local\Programs\Python\Python310\lib\site-packages\djitellopy\tello.py", line 579, in land
  File "C:\Users\gt\AppData\Local\Programs\Python\Python310\lib\site-packages\djitellopy\enforce_types.py", line 54, in wrapper
  File "C:\Users\gt\AppData\Local\Programs\Python\Python310\lib\site-packages\djitellopy\tello.py", line 486, in send_control_command
  File "C:\Users\gt\AppData\Local\Programs\Python\Python310\lib\site-packages\djitellopy\enforce_types.py", line 54, in wrapper
  File "C:\Users\gt\AppData\Local\Programs\Python\Python310\lib\site-packages\djitellopy\tello.py", line 528, in raise_result_error
Exception: Command 'land' was unsuccessful for 4 tries. Latest response:        'Aborting command 'land'. Did not receive a response after 7 seconds'

M4GNV5, 2023-04-10:
It looks like your drone does respond, but with an error message `error enter stop`.

Have you tried calibrating the drone? What firmware version and what drone are you using?

GreatenAnoymous, 2023-04-10:
> 

My firmware version was 01.04.35.01. After updating to 01.04.91.01, it works well now. Thank you very much.

closed 2023-04-10 by GreatenAnoymous