damiafuentes/DJITelloPy #182: "me.turn_motor_on()" and "me.turn_motor_off()" issue

Dtr1567, 2023-04-10:
my tello drone can't enter the cooling mode or exit, it says on console invalid command "motoron" i tried to update the firmware of the tello drone but it says latest firmware so that means my drone is on the latest firmware, i hope that bug fixes!

ubesingha92, 2023-06-05:
I think your drone is a normal Tello drone. This function does not work with my normal Tello, but it works with Tello Edu.

closed 2023-06-09 by M4GNV5

Dtr1567, 2023-07-26:
Thanks for help!

L7342, 2025-03-31:
> Like [@ubesingha92](https://github.com/ubesingha92) said, this is only available using the latest firmware on EDU variants

Do "set_vedio_fps/resolution" "send_keepalive" "send_expansion_command" etc (new commands added in SDK v3.0,mentioned in #92) are all available for Tello Edu&Talent (not the normal Tello)  only ? I face the similar problems in using these commands with normal Tello(version 01.04.92.01,app says it's the latest firmware)