damiafuentes/DJITelloPy #187: Issues with Tello video stream to Windows 11 Computer

lavisomers, 2023-05-27:
Hello everyone!

I am trying to follow along Murtaza's Workshop YouTube "Drone Programming With Python Course | 3 Hours | Including x4 Projects | Computer Vision" course:

https://www.youtube.com/watch?v=LmEcyQnfpDA

However I have hit a wall at the imageCapture section.

The Tello drone connects to the computer and I can use the basic movements commands without any difficulty, having the drone take off / move around with commands issued from the laptop. However when I try to run the code below to get a video stream:

```
from djitellopy import tello
import cv2

my_drone = tello.Tello()
my_drone.connect()
print(my_drone.get_battery())
my_drone.streamon()
frames = my_drone.get_frame_read()
img = frames.frame
img = cv2.resize(img, (360, 240))
cv2.imshow("Image", img)

```
When the code reached line 1029 in tello.py:
tello.cap = cv2.VideoCapture(address)

It got stuck for a while and then the tello drone blinked in red (instead of green) which caused the tello wifi to disconnect. 
Then I got the following output:

```
[INFO] tello.py - 122 - Tello instance was initialized. Host: '192.168.10.1'. Port: '8889'.
[INFO] tello.py - 437 - Send command: 'command'
[INFO] tello.py - 461 - Response command: 'ok'
41
[INFO] tello.py - 437 - Send command: 'streamon'
[INFO] tello.py - 461 - Response streamon: 'ok'
Traceback (most recent call last):
  File "C:\Users\User\bird\drone\drone_playground.py", line 27, in <module>
    frames = my_drone.get_frame_read()
  File "C:\Users\User\anaconda3\envs\drone\lib\site-packages\djitellopy\enforce_types.py", line 54, in wrapper
    return func(*args, **kwargs)
  File "C:\Users\User\anaconda3\envs\drone\lib\site-packages\djitellopy\tello.py", line 420, in get_frame_read
    self.background_frame_read = BackgroundFrameRead(self, address)  # also sets self.cap
  File "C:\Users\User\anaconda3\envs\drone\lib\site-packages\djitellopy\tello.py", line 1048, in __init__
    raise Exception('Failed to grab first frame from video stream')
Exception: Failed to grab first frame from video stream
[INFO] tello.py - 437 - Send command: 'streamoff'
[WARNING] tello.py - 447 - Aborting command 'streamoff'. Did not receive a response after 7 seconds
[INFO] tello.py - 437 - Send command: 'streamoff'
[WARNING] tello.py - 447 - Aborting command 'streamoff'. Did not receive a response after 7 seconds
[INFO] tello.py - 437 - Send command: 'streamoff'
[WARNING] tello.py - 447 - Aborting command 'streamoff'. Did not receive a response after 7 seconds
Exception ignored in: <function Tello.__del__ at 0x000002907F183F70>
Traceback (most recent call last):
  File "C:\Users\User\anaconda3\envs\drone\lib\site-packages\djitellopy\enforce_types.py", line 54, in wrapper
  File "C:\Users\User\anaconda3\envs\drone\lib\site-packages\djitellopy\tello.py", line 1019, in __del__
  File "C:\Users\User\anaconda3\envs\drone\lib\site-packages\djitellopy\enforce_types.py", line 54, in wrapper
  File "C:\Users\User\anaconda3\envs\drone\lib\site-packages\djitellopy\tello.py", line 1008, in end
  File "C:\Users\User\anaconda3\envs\drone\lib\site-packages\djitellopy\enforce_types.py", line 54, in wrapper
  File "C:\Users\User\anaconda3\envs\drone\lib\site-packages\djitellopy\tello.py", line 599, in streamoff
  File "C:\Users\User\anaconda3\envs\drone\lib\site-packages\djitellopy\enforce_types.py", line 54, in wrapper
  File "C:\Users\User\anaconda3\envs\drone\lib\site-packages\djitellopy\tello.py", line 486, in send_control_command
  File "C:\Users\User\anaconda3\envs\drone\lib\site-packages\djitellopy\enforce_types.py", line 54, in wrapper
  File "C:\Users\User\anaconda3\envs\drone\lib\site-packages\djitellopy\tello.py", line 528, in raise_result_error
Exception: Command 'streamoff' was unsuccessful for 4 tries. Latest response:	'Aborting command 'streamoff'. Did not receive a response after 7 seconds'

Process finished with exit code 1

```
Afterwards I wasn't able to reconnect to the tello wifi and I couldn't connect to any other wifi. After I restarted the computer I was able to connect to the wifi normally.
The issue isn't with the Tello drone - it can send a video to my phone.

In order to solve this issue I tried the following steps without success:
1. I created a thread that sends the command "command" to the drone every 3 seconds to keep it alive.
2. I tried to downgrade the opencv-python library to version 4.5.1.48 according to https://github.com/damiafuentes/DJITelloPy/issues/90#issuecomment-842336027 , but it didn't worked.
3. I disabled the windows firewall.
4. I updated the drone firmware to 1.04.92.01

Virtual environment installations:
djitellopy    2.4.0
numpy         1.24.3
opencv-python 4.5.1.48
pip           23.1.2
setuptools    67.7.2
wheel         0.40.0

Thank you!

DanMan32, 2023-05-28:
I have the same problem, running Windows 11.
Python version 3.10 - probably not compatible, but this Git says Python > 3.6!
I understand there's a problem with recent versions of OpenCV not able to read the Tello video stream, but I can't install n 4.5.1.48, all sorts of compatibility errors during the install.  Seems not compatible with 3.10.  Many of the dependencies this version of OpenCV uses aren't either.

I'd hate to have to downgrade my Python install.

ubesingha92, 2023-06-05:
use this code: 
from djitellopy import Tello
import cv2

my_drone = Tello()
my_drone.connect()
print(my_drone.get_battery())
my_drone.streamon()
while True:
    # Read a frame from the video stream
    img = my_drone.get_frame_read().frame
    img = cv2.resize(img, (360, 240))
    # Display the frame
    cv2.imshow("Image", img)
    # Check for key press to exit
    if cv2.waitKey(1) & 0xFF == ord('q'):
        break

# Clean up
cv2.destroyAllWindows()
my_drone.streamoff()
my_drone.land()

ccjimmy777, 2023-06-18:
I have the same problem. Is there any solution now?

lavisomers, 2023-06-18:
I managed to solve this issue by using a different PC with the same installation and Tello drone.

Kartik-Mehta, 2024-10-14:
I am unable to get the video stream due to same issue. Has it been resolved?

M4GNV5, 2024-10-15:
Could you provide some information on what python, djitellopy, etc. versions you are using? We moved away from opencv in favor of pyAV. I was assuming this issue to be resolved.

KartikMehta-1, 2024-10-15:
Hey, I was following Murtaza's workshop video: https://www.youtube.com/watch?v=LmEcyQnfpDA&list=PLMoSUbG1Q_r9p7iYBg6z6tZP002DAJ41H&index=4
and ended up here looking at the error message.

Python version: 3.11.3 & 3.7.0
djitellopy: 2.5.0
opencv: 4.10.0

Is there a recommended tutorial based on the latest version?

apezzi75, 2025-06-17:
Test this code. It works! (Windows 11, Python 3.12.10, Opencv 4.11

from threading import Thread
from djitellopy import Tello
import time, cv2, numpy
tello=Tello()

#connessione al drone
tello.connect()
print("Livello batteria: " + str(tello.get_battery()))
print("Temperatura interna: " + str(tello.get_temperature()))
keepRecording = True
tello.streamon()
video = cv2.VideoWriter('video.avi', cv2.VideoWriter_fourcc(*'XVID'), 30, (1080,740))

def videoRecorder():
    # create a VideoWrite object, recoring to ./video.avi
    i=0
    try:
      while keepRecording:
        img = tello.get_frame_read().frame
        img = cv2.resize(img, (1080, 740))
        video.write(img)
        time.sleep(1 / 30)
        print("Frame n."+str(i))
        i+=1
    except Exception as e:
      print("Errore video.release(): " + str(e))

# we need to run the recorder in a seperate thread, otherwise blocking options
try:
 recorder = Thread(target=videoRecorder)
 recorder.start()
 # tello.takeoff()
 time.sleep(5)
 # tello.land()
except Exception as e:
 print(str(e))
finally:
 keepRecording = False
 # cv2.destroyAllWindows()
 recorder.join()
 video.release()
 tello.streamoff()