damiafuentes/DJITelloPy #192: reading video stream with the python 3.10

SimaoJones, 2023-06-27:
Good afternoon,
My name is Simão Franco Jones, and I am using a DJI Ryze Tello EDU drone for a project.
I am programming the drone in Python 3.10 and the drone seems to be able to take off and moving.
My problem is that i cannot get the frames from the camera using the function get_frame_read().
I was reading the comments about this issue and they suggest to change the versions of the packages.
Is there another solution rather than changing or downgrading the versions of the packages?

SimaoJones, 2023-06-27:
I also I tried to use cv2.waitKey() and time.sleep() to give more time to get the frames and also it did not worked

jimmyfu0729, 2023-08-14:
I also encounter this problem, my version is python 3.11, there is no frame read from Tello even I run the examples without changes.

manavkulshrestha, 2025-04-28:
Any update on this?