damiafuentes/DJITelloPy #195: Failed to grab video frames from video stream

Watchkut, 2023-07-24:
I have tried multiple fixes suggested here but i cant get any captures or videos from my Tello Edu.

I get this error:  

Traceback (most recent call last):
  File "C:\Users\sanur\Desktop\New folder (2)\Tello-Object-Tracking\venv\lib\site-packages\djitellopy\tello.py", line 1049, in __init__
    self.container = av.open(self.address, timeout=(Tello.FRAME_GRAB_TIMEOUT, None))
  File "av\container\core.pyx", line 401, in av.container.core.open
  File "av\container\core.pyx", line 272, in av.container.core.Container.__cinit__
  File "av\container\core.pyx", line 292, in av.container.core.Container.err_check
  File "av\error.pyx", line 336, in av.error.err_check
av.error.ExitError: [Errno 1414092869] Immediate exit requested: 'udp://@0.0.0.0:11111'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\sanur\Desktop\New folder (2)\Tello-Object-Tracking\TelloMain.py", line 28, in <module>
    frame_read = me.get_frame_read()
  File "C:\Users\sanur\Desktop\New folder (2)\Tello-Object-Tracking\venv\lib\site-packages\djitellopy\enforce_types.py", line 54, in wrapper
    return func(*args, **kwargs)
  File "C:\Users\sanur\Desktop\New folder (2)\Tello-Object-Tracking\venv\lib\site-packages\djitellopy\tello.py", line 421, in get_frame_read
    self.background_frame_read = BackgroundFrameRead(self, address, with_queue, max_queue_len)
  File "C:\Users\sanur\Desktop\New folder (2)\Tello-Object-Tracking\venv\lib\site-packages\djitellopy\tello.py", line 1051, in __init__
    raise TelloException('Failed to grab video frames from video stream')

M4GNV5, 2023-08-01:
Is your tello connected to a Wifi in client/station mode?
You probably need to update your firmware.

Watchkut, 2023-08-01:
It was. I lowered my python version to 3.8 and it started working.

closed 2023-08-01 by Watchkut

TSoli, 2024-02-07:
> It was. I lowered my python version to 3.8 and it started working.

What exact version of Python did you downgrade to? I just tried downgrading to Python 3.8.18 and it did not work for me. I am on Linux Mint OS.