damiafuentes/DJITelloPy #200: Command of movement

Bitahsni, 2023-08-26:
Hello everyone. I have a problem! I want to move a tello drone under 20 cm to each direction, but the command of movement moves the drone from 20-100 cm.
Is there any way to move a drone under 20 cm??Is there an interrupt command?
Is it possible to command the drone to rotate with the desired angle?

M4GNV5, 2023-08-26:
By default the Tello is not able to make precise enough movements to move just 20cm. When commanding it to move 90cm it could move 95 or 85.

In order to improve accuracy you need a fixed point in space for orientation, which allows you to measure the 20cm. If you have a Tello EDU you can use a mission pad and the [go_xyz_speed_mid](https://djitellopy.readthedocs.io/en/latest/tello/#djitellopy.tello.Tello.go_xyz_speed_mid) function.

Regarding rotation: The functions you are looking for are: [rotate_clockwise](https://djitellopy.readthedocs.io/en/latest/tello/#djitellopy.tello.Tello.rotate_clockwise) and [rotate_counter_clockwise](https://djitellopy.readthedocs.io/en/latest/tello/#djitellopy.tello.Tello.rotate_counter_clockwise)

Bitahsni, 2023-08-28:
@M4GNV5  Thank you for your response. I don't have a Tello EDU, but my issue was solved with the send_rc_control()

closed 2023-08-29 by Bitahsni