damiafuentes/DJITelloPy #202: DJITelloPy in Google Colab

camchr, 2023-09-11:
Greetings. I am trying to use DJITelloPy in Google Colab. Because my Tello connects locally via WiFi to my computer, it is not possible for the Google server to recognize it. How can I make it work? Thank you.

M4GNV5, 2023-10-09:
There probably isnt a very easy solution. In order for google colab (or any cloud hosted interface) to connect to your Tello you have to make your tello accessible over the internet. On Tello EDUs in station mode you could probably configure a port forwarding in your router. With a regular tello and a computer connected both to the tello and to the internet forwarding is more complex.
You might be able to set something up using firewall rules and/or a service like ngrok.io, but this isn't a beginners task. You should probably look into routing the UDP traffic from a publicly accesible endpoint to your Tello.

Nevertheless this is a network topic and not really a Tello or DJITelloPy specific issue.

camchr, 2023-10-09:
Greetings. Thanks for the reply. I'm going to try what you suggest.
Thank you so much.