damiafuentes/DJITelloPy #217: non-existing PPS 0 referenced no frame error

semikernel, 2024-03-20:
My code is simple as followings:
```py 
from djitellopy import tello
import cv2
import time

me = tello.Tello()
me.connect()
me.streamon()

print(me.get_battery())
time.sleep(5)

while True:

    img = me.get_frame_read().frame
    img = cv2.resize(img,(360,240))
    cv2.imshow("Image",img)
    cv2.waitKey(100)
```

However, I didn't get a video stream from tello drone in my computer, and the terminal output log is :
```bash
INFO] tello.py - 130 - Tello instance was initialized. Host: '192.168.10.1'. Port: '8889'.
[INFO] tello.py - 439 - Send command: 'command'
[INFO] tello.py - 463 - Response command: 'ok'
[INFO] tello.py - 439 - Send command: 'streamon'
[INFO] tello.py - 463 - Response streamon: 'ok'
68
non-existing PPS 0 referenced
non-existing PPS 0 referenced
decode_slice_header error
no frame!
```

I tried a lot of time but still not woked. So how could I fix that problem?

yam1917, 2024-03-23:
same issue

CastleChen339, 2024-03-25:
Wait for a while, it will take some time for the camera to turn on