damiafuentes/DJITelloPy #218: Issue connect to wifi

yzysnake, 2024-04-15:
Hi, I constantly have issues connecting the drone to my home wifi which offers both 5 GHz and 2.4 GHz. Any idea to fix it by chance?

yzysnake, 2024-04-15:
I could connect the drone to other routers, for example, a Google Nest Wi-Fi or any other hot spots. My home router is an ASUS RT-AX3000.

Nekomeomeo, 2024-05-19:
Hi,Can U teach me how to change the default wifi(like TELLO-V2LC7G) into a  hot spots? I use the function in djitellopy :"connect_to_wifi(ssid,password)"to change the target router,but it seems not work.Can U share your experience?

M4GNV5, 2025-03-13:
@yzysnake most routers support setting separate names for 2.4G vs 5G. Can you try this to verify it is a 5Ghz coexistence issue?

@Nekomeomeo this is only supported by Tello EDUs running the latest firmware. If you are getting an `unknown command` error try updating your drone.