damiafuentes/DJITelloPy #219: Cannot set bitrate, fps and resolution

kiloluke, 2024-05-05:
Having an issue setting the bitrate, fps and resolution

`me = tello.Tello()
# me.set_video_fps(Tello.FPS_15)  #fails
me.connect()
print(me.get_battery())
me.takeoff()
me.send_rc_control(0, 0, 25, 0)
me.set_speed(100)
me.set_video_bitrate(Tello.BITRATE_2MBPS)
me.streamon()`

Tried moving around the commands, did not help. djitellopy.tello.TelloException: Command 'setbitrate 2' was unsuccessful for 4 tries. Latest response:	'unknown command: setbitrate'

M4GNV5, 2024-05-06:
Which Tello model are you using? Try updating the firmware using the official app.

kiloluke, 2024-05-06:
> Which Tello model are you using? Try updating the firmware using the official app.

The firmware is updated, the drone model is Tello Boost Combo