damiafuentes/DJITelloPy #220: Not able to read Dji tello frames on ubuntu 20.04

Vaibhav1196, 2024-05-12:
This is my code, I have also tried adding delays but tit never worked, I received the same result :

############################################################################################

import cv2
from djitellopy import Tello
import time

# Initialize Tello and set up video stream
tello = Tello()
tello.connect()
tello.streamon()

time.sleep(5)

frame_read = tello.get_frame_read()

time.sleep(5)

# Main loop to continuously read and display frames
while True:
    # Attempt to get a new frame from the frame queue
    frame = frame_read.frame
    
    # Check if the frame is not None
    if frame is not None:
        # Display the frame using OpenCV
        cv2.imshow("Tello Video", frame)
    
    # Check for key press to exit the loop
    if cv2.waitKey(1) & 0xFF == ord('q'):
        break

# Clean up
cv2.destroyAllWindows()
tello.streamoff()
tello.end()

############################################################################################

I use Anaconda environment with : 

opencv 4.9
djitellopy latest version

############################################################################################

ERROR : 

[INFO] tello.py - 129 - Tello instance was initialized. Host: '192.168.10.1'. Port: '8889'.
[INFO] tello.py - 438 - Send command: 'command'
[INFO] tello.py - 462 - Response command: 'ok'
[INFO] tello.py - 438 - Send command: 'streamon'
[INFO] tello.py - 462 - Response streamon: 'ok'
Traceback (most recent call last):
  File "/home/vaibhav/anaconda3/envs/tello/lib/python3.8/site-packages/djitellopy/tello.py", line 1049, in __init__
    self.container = av.open(self.address, timeout=(Tello.FRAME_GRAB_TIMEOUT, None))
  File "av/container/core.pyx", line 420, in av.container.core.open
  File "av/container/core.pyx", line 266, in av.container.core.Container.__cinit__
  File "av/container/core.pyx", line 286, in av.container.core.Container.err_check
  File "av/error.pyx", line 328, in av.error.err_check
av.error.ExitError: [Errno 1414092869] Immediate exit requested: 'udp://@0.0.0.0:11111'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./record_tello.py", line 46, in <module>
    frame_read = tello.get_frame_read()
  File "/home/vaibhav/anaconda3/envs/tello/lib/python3.8/site-packages/djitellopy/enforce_types.py", line 54, in wrapper

keecharles, 2024-07-10:
djitellopy 2.5.0 use pyav to decode the stream, pyav and opencv can output the frame well on windows, but has conflict on ubuntu 20.04. 
Try pip install djitellopy==2.4.0 on ubuntu.

ZenroChia, 2025-07-23:
Did  installing djitellopy==2.4.0 work?