damiafuentes/DJITelloPy #221: OpenCV is not loading while imported the djitellopy library

Jere2k03, 2024-06-10:
I want to display the live cam of my tello with OpenCV but it only shows errors. I use a Raspberry Pi 5
So I tried a test function just for openCV to see if everything works correctly with cv2.
so my test code is:
import cv2

def main(args):
    image = cv2.imread('testcv2.png')
    if image is None:
        print("pic wrong")
        return
    cv2.imshow("Test", image)
    cv2.waitKey(0)
    cv2.destroyAllWindows()

if __name__ == '__main__':
    import sys
    sys.exit(main(sys.argv))
    
With this code my picture normally opens with openCV.
But if i add `from djitellopy import Tello` at the beginning, the code runs without an error but opencv does not open the picture again.

Can sb help me pls?

freeutka-cmd, 2024-06-20:
```
import cv2
from djitellopy import Tello

def main():
    # Initialize Tello
    tello = Tello()
    tello.connect()
    tello.streamon()

    try:
        while True:
            frame = tello.get_frame_read().frame
            cv2.imshow("Tello Live Stream", frame)

            if cv2.waitKey(1) & 0xFF == ord('q'):
                break
    finally:
        tello.streamoff()
        cv2.destroyAllWindows()

if __name__ == '__main__':
    main()
```

freeutka-cmd, 2024-06-20:
Try this, idk

closed 2024-06-24 by Jere2k03