damiafuentes/DJITelloPy #231: send_rc_control seems to be bugged

Xyneef, 2024-11-05:
I am encountering an issue on a DJI Tello, using Python 3.13 on the latest PyCharm IDE and the djitellopy library version 2.5.0 installed via PYCharm's package manager.
When sending velocity commands via the _send_rc_control()_ method, I noticed that the _up_down_velocity_ parameter is actually causing the drone to strafe left and right, causing some inconvenience to say the least. It's as if the _up_down_ channel is being interpreted as the _left_right_ channel.  
Is this happening only to me?

M4GNV5, 2024-11-05:
Can you provide a log of the program you are running. This way we might be able to narrow down wether it is a problem with the code or your drone.

Xyneef, 2024-11-07:
Hi, thanks for the reply. Allow me to share both my code and the log. 
For some reason I do not seem to understand, this code had been completely failing at doing what it's supposed to do (i.e., just lifting the drone up , rotating it clockwise, then bring it down until it automatically lands).
It was instead sending the drone crashing into the walls immediately after the takeoff. 
Now, for some other reason I also do not seem to understand, when I ran this 5 minutes ago to copy-paste the log, it worked flawlessly....

 ` 
 from djitellopy import Tello
 import time

 def height(i):

     for seconds in range(i):
         print("Height Level: %i cm" %(me.get_height()))
         time.sleep(1)
     return

 me = Tello()
 me.connect()

 print("Battery Level: %i %%" %(me.get_battery()))
 me.takeoff()
 time.sleep(4)
 me.send_rc_control(0,0, 50,0)
 height(4)
 me.send_rc_control(0, 0, 0, 40)
 height(4)
 me.send_rc_control(0,0,-10,0)
 height(4)

 me.end() 

`

The log that I ran 5 minutes ago is the following :

[INFO] tello.py - 129 - Tello instance was initialized. Host: '192.168.10.1'. Port: '8889'.
[INFO] tello.py - 438 - Send command: 'command'
[INFO] tello.py - 462 - Response command: 'ok'
Battery Level: 60 %
[INFO] tello.py - 438 - Send command: 'takeoff'
[INFO] tello.py - 462 - Response takeoff: 'ok'
Height Level: 70 cm
[INFO] tello.py - 471 - Send command (no response expected): 'rc 0 0 50 0'
Height Level: 70 cm
Height Level: 80 cm
Height Level: 110 cm
Height Level: 150 cm
[INFO] tello.py - 471 - Send command (no response expected): 'rc 0 0 0 40'
Height Level: 170 cm
Height Level: 170 cm
Height Level: 160 cm
Height Level: 160 cm
[INFO] tello.py - 471 - Send command (no response expected): 'rc 0 0 -10 0'
Height Level: 170 cm
Height Level: 170 cm
Height Level: 160 cm
[INFO] tello.py - 438 - Send command: 'land'
[INFO] tello.py - 462 - Response land: 'ok'

Process finished with exit code 0

M4GNV5, 2024-11-13:
There are a lot of aspects influencing the drones. The biggest one is lighting: Maybe second time lighting was different / brighter sun / turned off.
It generally helps calibrating the drone (via the official app).
Maybe you experienced an unlucky packet loss between drone and computer (i do not think rc_control is resend).

I will be closing this, since the issue seems to be resolved.

closed 2024-11-13 by M4GNV5