damiafuentes/DJITelloPy #235: Running on Android and other supported OS

iamjonny, 2025-01-30:
I've started to play with python using pydroid3. Also have a tello, and found the tellopy libraries use numpy, which won't work for me on android12.

Gpt suggested running on Ubuntu **on android** but although the Ubuntu install seemed to work via termux, the subsequent pkg/pip install attempts failed.

So maybe a computer is needed to run the python on, as oppose to developing and running exclusively on android, even without windows/Mac/Linux machine to run the program on.

Alternatively if there's a VM that can run on the android that lets you use "full python".

Nb I found that the tellopy_reduced library installs ok, but not sure what differences there are in usage, or what limitations that will introduce.

Do you need a computer in the tech stack to do tello python stuff, or is there any way to do on Just android?

Maybe earlier androids allowed python to do more on them?

Thanks!

M4GNV5, 2025-01-31:
currently numpy is only used for handling video stream frames. If you do not intend to use video streaming you could probably still use this on android.

Looking at the code it does not seem like numpy is really required, it is only used for converting AV frames to numpy arrays (probably for legacy reasons). Feel free to create a PR for removing numpy as a dependency alltogether.

I would not be too sure if `AV` and/or `opencv` are available on android though.

ar-cyber, 2025-03-13:
> currently numpy is only used for handling video stream frames. If you do not intend to use video streaming you could probably still use this on android.
>
> Looking at the code it does not seem like [...]

It would work but it would probably be too slow to use.