damiafuentes/DJITelloPy #239: Possibility and problems for sending commands without return

dvdmc, 2025-04-02:
Thanks for your work! I was wondering if it's possible to send the movement commands without return to override previous commands. What problems might arise? If there are none, could the library add this option?

M4GNV5, 2025-04-03:
I do not think the drone accepts movement commands while already handling one.
The `send_rc_control` function however uses sending without return. This one can be used to continuously update. Is this what you are looking for?

https://djitellopy.readthedocs.io/en/latest/tello/#djitellopy.tello.Tello.send_rc_control

dvdmc, 2025-04-03:
Yes, that would be my approach. I was wondering if someone already tried before or has any insights on it. Since it's not currently possible to do it in the library, I thought that probably there is something on the low level that prevents it, as you mentioned.